JonasVautherin/px4-gazebo-headless #11: Problem of connecting with mavsdk on the remote computer

alexey-gavrilov, 2020-07-13:
I tried connect px4-gazebo-headless  using its docker with mavsdk example program (takeoff_and_land) that located on remote computer.

I set docker ip equal to 172.17.0.3/16, host computer ip equal to 10.1.0.9/8, remote computer ip equal to 10.1.2.143/8

I used next command to run take_off_land: 
./takeof_and_land udp://:14540

At first I used  next command to connect docker with host computer:
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.10.1
All worked fine.

Then I used next command to connect docker with remote computer:
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.10.1 10.1.2.143
But in this case the message “No system found , exiting” fired!

So how can I decide this problem?? What I did wrong?

alexey-gavrilov, 2020-07-13:
Because I couldn't see any udp message from tcpdump , I also have tried to use the mavlink-routerd software.
On the remote computer i executed next command:
mavlink-routerd  -e 10.1.2.143:14540 127.0.0.1:14540
Then I saw next udp  messages with tcpdump:
IP 10.1.0.9 .14580 > 10.1.2.143:14540, length 36
IP 10.1.0.9 .14580 > 10.1.2.143:14540, length 29
IP 10.1.0.9 .14580 > 10.1.2.143:14540, length 60
P 10.1.0.9 .14580 > 10.1.2.143:14540, length 36
IP 10.1.0.9 .14580 > 10.1.2.143:14540, length 28
and so on
But I any way couldnt connect with takeoff_and_land with simulator!

julianoes, 2020-07-14:
I just tried this and it worked for me.

Can you paste what output you get for the case that doesn't work?

I get this:
```
pxh> INFO  [mavlink] using network interface eth0, IP: 172.17.0.2
INFO  [mavlink] with netmask: 255.255.0.0
INFO  [mavlink] and broadcast IP: 172.17.255.255
INFO  [ekf2] Mag sensor ID changed to 197388
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] 728000: EKF GPS checks passed (WGS-84 origin set)
INFO  [mavlink] using network interface eth0, IP: 172.17.0.2
INFO  [mavlink] with netmask: 255.255.0.0
INFO  [mavlink] and broadcast IP: 172.17.255.255
```

tvp100, 2022-04-26:
I also had the problem. I can only connet the localhost device. (And I must assign the ip to 192.168.220.1, which is my computer IP)
`await drone.connect(system_address="udp://192.168.220.20:14540")`
This code will not work.
Only replact the 192.168.220.20 to localhost or omit it can work.
Is there any way to connect to a custom IP?

julianoes, 2022-05-03:
@tvp100 the drone needs to send it to where MAVSDK is running, not the other way round.

So PX4 needs to broadcast or send it to an IP, see:
https://docs.px4.io/master/en/simulation/#enable-udp-broadcasting
https://docs.px4.io/master/en/simulation/#enable-streaming-to-specific-address

closed 2023-12-04 by JonasVautherin