JonasVautherin/px4-gazebo-headless #13: How to find the clients connected with MAVSDK Server?

MuhammadBilal1, 2020-07-20:
I am running the [Gazebo Headless Simulator](https://github.com/jonasvautherin/px4-gazebo-headless#px4-gazebo-simulator-headless) with the below command.

`docker run --rm -it jonasvautherin/px4-gazebo-headless:1.10.1`

After successfully running the simulator, I tried to ran the [example](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/mission.py). 
`python mission.py` This example ran successfully.

But when I try to ran the mission example from my own application (basically a Django application with Celery for running MAVSD in background) If I get any problem while running the example, after the error I am unable to connect with the Drone (Gazebo Headless Simulator). I tried to stop the simulator and started again, but still not working.

1- How to check is there any client connected with MAVLINK?
2- If an error came while running the example and program failed without completing the "Landing", the problem didn't able to connect with the drone after the error.

[image]

MuhammadBilal1, 2020-07-20:
@julianoes @JonasVautherin  Can you please help me with the above problem.

JonasVautherin, 2020-07-20:
I'm not sure what kind of error you are referring to. You mean that `mission.py` crashes?

MuhammadBilal1, 2020-07-21:
Error is appearing while executing the mission.py from my Django application via Celery.

Error is coming while uploading the mission to the connected drone. Now, after getting this error, I can't connect to the drone again until I restart my Operating System, or I close all my terminal windows, which are running Django/Celery/Gazebo HeadLess Simulator.

**Problems:**
1- Getting the `Operation Timeout error` at `upload_mission()` as shown in the below image while running the mission.py.
2- After getting the operation timeout error, mission.py remains stuck at `Waiting for Drone to connect...`.

[image]

MuhammadBilal1, 2020-07-21:
Somehow I managed to get the list of ports that were in use when I am unable to connect with the Drone. 
Please see the attached snapshot for details.
**I need help in resolving the Operation Timeout error of mission.py**

[image]

JonasVautherin, 2020-07-21:
Maybe you should try to run `mavsdk_server` manually, with:

```
./mavsdk_server -p 50051 udp://:14540
```

And in Python, you need to connect to it by changing the `System()` line:

```
drone = System(mavsdk_server_address='localhost', port=50051)
```

MuhammadBilal1, 2020-07-21:
I am unable to start the `mavsdk_server` manually, can you please tell me from where I need to run this command?
Please see the attached snapshot.

[image]

julianoes, 2020-07-22:
You can look for it:
```
find . -name mavsdk_server
```

For me, the one installed using pip is here:
```
~/.local/lib/python3.8/site-packages/mavsdk/bin/mavsdk_server
```

MuhammadBilal1, 2020-07-22:
@julianoes  Thank you for your support.
I installed the mavsdk using the pip, and my installation is also available at the same place. I went to that directory and started the mavsdk_server manually. The server started successfully.

**My issue is resolved and I am not getting the `Operation timeout` error anymore while running the `mission.py` example.**

I started the things in the following order, and everything worked fine.

1- Started mavsdk_server manually from your given command. Ran successfully.
2- Started the Gazebo Headless simulator. Ran successfully.
3- Started by application (Django). Ran successfully.
4- Started Celery. Ran successfully.
5- Triggered the method which was calling the `mission.py`. It also ran successfully.

[image]

closed 2020-07-22 by MuhammadBilal1