JonasVautherin/px4-gazebo-headless #16: Anyway to broadcast to the secondary mavlink device (phone or second computer) instead of using specific IP of device?

UASOP, 2020-08-29:
Hello @JonasVautherin , I was wondering if there is a way to broadcast the ip address for a second computer on the network to connect to the docker container? This would be without specifying the device IP as described in the instructions. Thanks in advance for the help.

JonasVautherin, 2020-08-30:
I am not sure I understand what you mean. Right now you can give the IP to which 14540 must be sent (presumably for a MAVSDK client), and the IP to which 14550 must be sent (that would be for QGC). Those can be two different computers on the network.

You won't be able to broadcast from the container to the network (broadcast messages don't reach the host). So you need to specify the IP. But that can be a phone or a second computer, of course.

UASOP, 2020-08-30:
Thanks @JonasVautherin. So in order to get the container to broadcast on the local network in order for any computer to connect to it, without needing to specify the specific IP address of the device trying to connect to it. Could MAVSDK server be used to broadcast it over the local network if it is running on the host computer of the container?

UASOP, 2020-08-30:
Quick update on my end @JonasVautherin I found a way to get the docker container to broadcast n the local network through a third party.  For those who may be trying to do the same thing: The third party is MAVP2P. I am using the (udps:ipaddress and port here) to forward the address to on the host computer to the local network. QGC on any device can be connect to the simulated aircraft within the container as long as they have access to the host computers IP address and are on the same network. With no other point to point internal comms security getting in the way.so using a home wifi or something similar should work
.

UASOP, 2020-08-30:
Closing this issue

closed 2020-08-30 by UASOP