JonasVautherin/px4-gazebo-headless #24: Issue Running Video Stream

ekwilinski, 2022-05-27:
Hello,

I am having trouble connecting to the video stream of the typhoon h480. I ran the command in the readme to get a video feed, but when I try to connect to localhost:8554 in vncviewer, nothing happens. 

Thank you,
Elliot

Sitin, 2022-06-29:
I experience the same problem. It seems that the stream should be available at `tsp://127.0.0.1/live` but my QGroundControl fails to get any video.

Sitin, 2022-07-09:
@ekwilinski I can open video stream with (you need to install [VLC](https://www.videolan.org/vlc/)):

```shell
vlc rtsp://127.0.0.1:8554/live 
```

Still can't configure QGroundControl to receive this stream.

julianoes, 2022-07-11:
The problem might be that PX4 automatically sends the video stream parameters via mavlink but since there is an additional RTSP server this does not actually work as expected.

It's as if there should be a setting named "Auto MAVLink" in QGC and then a setting to manually set it.

stevehenderson, 2022-11-12:
I was able to enter the uri in @Sitin vlc start cmd directly into QGC:

[image]
[image]

closed 2025-01-15 by JonasVautherin