JonasVautherin/px4-gazebo-headless #26: Publish image for `arm64`

Sitin, 2022-06-27:
At the moment it is not possible to run the existing image on `arm64` and OS X due to a qemu bug.

Fortunately I've just successfully built an image for `amd64` with one minor amendment:

```
RUN pip3 install --upgrade pip && \   # pip should be upgraded to support multiple patforms
    pip3 install empy \
    ...
```

It would be helpful if `arm64` image will be published on Docker Hub. It can be done by a [simple change](https://github.com/marketplace/actions/run-on-architecture) in GitHub pipelines config.

julianoes, 2022-06-27:
I'd suggest to make a PR here that adds what you need to CI.

Sitin, 2022-06-28:
I've added PR: #27.

julianoes, 2022-06-30:
Thanks @Sitin.

PR #27 (2022-07-04): ARM64 builds
Adds multi arch builds according to [this](https://dev.to/cloudx/multi-arch-docker-images-the-easy-way-with-github-actions-4k54) (corresponds to official Docker suggestions for QUEMU/buildx).

Additionally fixes:
- Main branch of `PX4-Autopilot` is now `main`, not `master`
- Upgrades PIP (prerequisite for multi-arch builds)

Resolves #26
closed 2022-07-04 by JonasVautherin