JonasVautherin/px4-gazebo-headless #31: Simulation error - Primary accelerometer not found

elpiel, 2022-08-07:
I'm pretty sure that this is an issue with the image itself. Prior to `1.11` (i.e. 0.10.1) this issue is not present:

```
px4-gazebo-headless    | WARN  [PreFlightCheck] Preflight Fail: Accel Range, hold still on arming
px4-gazebo-headless    | WARN  [PreFlightCheck] Primary accelerometer not found
px4-gazebo-headless    | WARN  [commander] Takeoff denied! Please disarm and retry
```

PX4 logs:
```
px4-gazebo-headless    | 
px4-gazebo-headless    | ______  __   __    ___ 
px4-gazebo-headless    | | ___ \ \ \ / /   /   |
px4-gazebo-headless    | | |_/ /  \ V /   / /| |
px4-gazebo-headless    | |  __/   /   \  / /_| |
px4-gazebo-headless    | | |     / /^\ \ \___  |
px4-gazebo-headless    | \_|     \/   \/     |_/
px4-gazebo-headless    | 
px4-gazebo-headless    | px4 starting.
px4-gazebo-headless    | 
px4-gazebo-headless    | INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
px4-gazebo-headless    | INFO  [param] selected parameter default file eeprom/parameters_10016
px4-gazebo-headless    | [param] parameter file not found, creating eeprom/parameters_10016
px4-gazebo-headless    |   SYS_AUTOCONFIG: curr: 0 -> new: 1
px4-gazebo-headless    | * SYS_AUTOSTART: curr: 0 -> new: 10016
px4-gazebo-headless    |   BAT_N_CELLS: curr: 0 -> new: 4
px4-gazebo-headless    |   CAL_ACC0_ID: curr: 0 -> new: 1311244
px4-gazebo-headless    |   CAL_ACC_PRIME: curr: 0 -> new: 1311244
px4-gazebo-headless    |   CAL_GYRO0_ID: curr: 0 -> new: 1311244
px4-gazebo-headless    |   CAL_GYRO_PRIME: curr: 0 -> new: 1311244
px4-gazebo-headless    |   CAL_MAG0_ID: curr: 0 -> new: 197388
px4-gazebo-headless    |   CAL_MAG_PRIME: curr: 0 -> new: 197388
px4-gazebo-headless    |   CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
px4-gazebo-headless    |   COM_RC_IN_MODE: curr: 0 -> new: 1
px4-gazebo-headless    |   EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
px4-gazebo-headless    |   EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
px4-gazebo-headless    |   COM_ARM_EKF_AB: curr: 0.0022 -> new: 0.0050
px4-gazebo-headless    |   EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
px4-gazebo-headless    |   MC_PITCH_P: curr: 6.5000 -> new: 6.0000
px4-gazebo-headless    |   MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
px4-gazebo-headless    |   MC_ROLL_P: curr: 6.5000 -> new: 6.0000
px4-gazebo-headless    |   MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
px4-gazebo-headless    |   MPC_Z_VEL_P_ACC: curr: 4.0000 -> new: 12.0000
px4-gazebo-headless    |   MPC_Z_VEL_I_ACC: curr: 2.0000 -> new: 3.0000
px4-gazebo-headless    |   MPC_XY_P: curr: 0.9500 -> new: 0.8000
px4-gazebo-headless    |   MPC_XY_VEL_P_ACC: curr: 1.8000 -> new: 4.0000
px4-gazebo-headless    |   MPC_XY_VEL_D_ACC: curr: 0.2000 -> new: 0.3200
px4-gazebo-headless    |   RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
px4-gazebo-headless    |   SDLOG_MODE: curr: 0 -> new: 1
px4-gazebo-headless    |   SDLOG_PROFILE: curr: 1 -> new: 131
px4-gazebo-headless    |   SDLOG_DIRS_MAX: curr: 0 -> new: 7
px4-gazebo-headless    |   SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
px4-gazebo-headless    |   SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
px4-gazebo-headless    |   TRIG_INTERFACE: curr: 4 -> new: 3
px4-gazebo-headless    |   COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
px4-gazebo-headless    |   IMU_INTEG_RATE: curr: 200 -> new: 250
px4-gazebo-headless    |   NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
px4-gazebo-headless    |   RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
px4-gazebo-headless    |   RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
px4-gazebo-headless    |   PWM_MAX: curr: 2000 -> new: 1950
px4-gazebo-headless    |   PWM_MIN: curr: 1000 -> new: 1075
px4-gazebo-headless    |   GPS_UBX_DYNMODEL: curr: 7 -> new: 6
px4-gazebo-headless    | * SYS_AUTOCONFIG: curr: 1 -> new: 0
px4-gazebo-headless    | INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
px4-gazebo-headless    | INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
px4-gazebo-headless    | INFO  [simulator] Simulator connected on TCP port 4560.
px4-gazebo-headless    | INFO  [commander] LED: open /dev/led0 failed (22)
px4-gazebo-headless    | INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
px4-gazebo-headless    | INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
px4-gazebo-headless    | INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14580 remote port 14540
px4-gazebo-headless    | INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
px4-gazebo-headless    | INFO  [logger] logger started (mode=all)
px4-gazebo-headless    | INFO  [logger] Start file log (type: full)
px4-gazebo-headless    | INFO  [logger] [logger] ./log/2022-08-07/13_23_01.ulg
px4-gazebo-headless    | INFO  [logger] Opened full log file: ./log/2022-08-07/13_23_01.ulg
px4-gazebo-headless    | INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
px4-gazebo-headless    | INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
px4-gazebo-headless    |   MAV_BROADCAST: curr: 0 -> new: 1
px4-gazebo-headless    | INFO  [px4] Startup script returned successfully
px4-gazebo-headless    | pxh> INFO  [mavlink] using network interface eth0, IP: 172.23.0.2
px4-gazebo-headless    | INFO  [mavlink] with netmask: 255.255.0.0
px4-gazebo-headless    | INFO  [mavlink] and broadcast IP: 172.23.255.255
px4-gazebo-headless    | INFO  [mavlink] using network interface eth0, IP: 172.23.0.2
px4-gazebo-headless    | INFO  [mavlink] with netmask: 255.255.0.0
px4-gazebo-headless    | INFO  [mavlink] and broadcast IP: 172.23.255.255
px4-gazebo-headless    | INFO  [ecl/EKF] 776000: GPS checks passed
px4-gazebo-headless    | INFO  [mavlink] partner IP: 172.23.0.3
px4-gazebo-headless    | INFO  [ekf2] Mag sensor ID changed to 197388
px4-gazebo-headless    | WARN  [PreFlightCheck] Preflight Fail: Accel Range, hold still on arming
px4-gazebo-headless    | WARN  [PreFlightCheck] Primary accelerometer not found
px4-gazebo-headless    | WARN  [commander] Takeoff denied! Please disarm and retry
```

julianoes, 2022-08-08:
That's odd. Looks like a PX4 issue. Although v1.11 is fairly old. Can't you move to v1.13?

elpiel, 2022-08-08:
@julianoes I think you did not understand me correctly - everything after `0.10.1` gives this error, including:
-  `1.11` (tested)
- `1.12` (I think I've tested this one too)
- `1.13` (tested):

```
px4-gazebo-headless    | WARN  [PreFlightCheck] Preflight Fail: Accel Range, hold still on arming
px4-gazebo-headless    | WARN  [PreFlightCheck] Primary accelerometer not found
```

It might be due to the fact that the Docker's entrypoint changed allowing vehicle and world?!

julianoes, 2022-08-09:
Hm, so is the change that breaks it one related to PX4 or MAVSDK? It looks like it would be PX4 but from your testing it seems to be related to MAVSDK?

elpiel, 2022-08-09:
I think it might be related to the image. It seems that the broadcasting changed and that's why mavsdk_server does not find the system. When I run mavsdk_server locally (without docker-compose) it finds the system and connects.

I tried different solutions like setting a custom network with static IPs:

```yml
    networks:
      internal-ip:
          ipv4_address: 172.20.0.6 # and 172.20.0.5 for px4-gazebo-headless
networks:
  internal-ip:
    driver: bridge
    ipam:
      driver: default
      config:
      - subnet: 172.20.0.0/24
```

using the services name (for mavsdk server) - `udp://px4-gazebo-headless:14540` but still nothing...

It does not find any system:
```bash
mavsdk-grpc-server     | [09:32:08|Info ] MAVSDK version: v1.4.4 (mavsdk_impl.cpp:20)
mavsdk-grpc-server     | [09:32:08|Info ] Waiting to discover system on udp://px4-gazebo-headless:14540... (connection_initiator.h:20)
mavsdk-grpc-server     | [09:32:08|Debug] Initializing connection to remote system... (mavsdk_impl.cpp:480)
```

elpiel, 2022-08-09:
Ok, so, the initial problem I had with `Primary accelerometer not found` was caused by a missing world - baylands.

I've renamed this issue because it is still an issue that I should be fixed and I'm opening a new issue to track down why `docker-compose` running `mavsdk_server` can't discover mavlink after updating to versions `> 0.10.1`

JonasVautherin, 2022-08-15:
Running
```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.0 -w baylands
```

I get:

```
Error Code: 11 Msg: Unable to find uri[model://baylands]
```

Not sure why, though. Where should the file be?

elpiel, 2022-08-15:
According to the docs (https://docs.px4.io/main/en/simulation/gazebo_worlds.html) the files are located in this repo:

https://github.com/PX4/PX4-SITL_gazebo/tree/main/worlds

Not sure where and how they are pulled in the PX4-Autopilot repo and SITL configuration though.

JonasVautherin, 2022-08-16:
Hmm, not sure why that changed. Please let me know if you find the root cause :blush: