JonasVautherin/px4-gazebo-headless #33: Docker run fails: px4-rc.simulator: ign: not found

av-jeroen, 2022-08-25:
Steps to reproduce:
 - `docker build -t px4-gazebo-headless .`
 - `docker run --rm -it px4-gazebo-headless`

The build is successful. The docker-run command fails and results in the following log:

```
...(omitted)...
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4_SIM_MODEL: iris
PX4_SIM_WORLD: empty
etc/init.d-posix/rcS: 10: /root/Firmware/build/px4_sitl_default/tmp/rootfs/etc/init.d-posix/px4-rc.simulator: ign: not found
ERROR [px4] Startup script returned with return value: 32512
[100%] Built target gazebo_iris__empty
```

The problem does not occur when directly pulling from Docker hub: `docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.0`

I did a manual build last week and that worked just fine. There are no changes on this repo, and the PX4 gazebo build works just fine (`make px4_sitl gazebo`). So not sure where something changed that is affecting this?

JonasVautherin, 2022-08-25:
By default it's building the [`main`](https://github.com/JonasVautherin/px4-gazebo-headless/blob/master/Dockerfile#L49) branch, which apparently needs some changes :+1:.

av-jeroen, 2022-08-25:
I did try to clone the stable release of PX4, still got the same error. 

```
# In Dockerfile
RUN git clone --branch v1.13.0 https://github.com/PX4/PX4-Autopilot.git ${FIRMWARE_DIR}
```

JonasVautherin, 2022-08-25:
Hmm I actually tried your exact commands, and it just worked for me:

```
docker build -t px4-gazebo-headless .
docker run --rm -it px4-gazebo-headless
```

av-jeroen, 2022-08-25:
I did a full rebuild and now it also works again for me. I tried it several times yesterday and today. Maybe a linux-package was broken? 

Anyways. Guess this is resolved now. Thanks for checking.

closed 2022-08-25 by av-jeroen