JonasVautherin/px4-gazebo-headless #34: docker run error

mengchaoheng, 2022-10-23:
```Console
parallels@ubuntu-linux-20-04-desktop:~$ docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.1
[  0%] Built target ver_gen
[  0%] Generating ../../logs
[  0%] Built target git_gazebo
[  0%] Built target parameters_xml
[  0%] Built target logs_symlink
[  0%] Built target events_header
[  0%] Built target flighttasks_generated
[  0%] Built target uorb_headers
[  0%] Built target actuators_json
[  0%] Built target drivers_board
[  0%] Built target mixer_gen_6dof
[  0%] Built target mixer_gen
[  0%] Built target output_functions_header
[  0%] Built target version
[  0%] Built target wind_estimator
[  0%] Built target world_magnetic_model
[  0%] Built target git_gps_devices
[  0%] Built target git_mavlink_v2
[  0%] Performing build step for 'sitl_gazebo'
make[5]: warning: -jN forced in submake: disabling jobserver mode.
[  0%] Built target events_json
[  0%] Built target tinybson
[  0%] Built target perf
[  2%] Built target work_queue
[  2%] Built target geo
[  2%] Built target MixerBase
[  8%] Built target std_msgs
[ 19%] Built target sensor_msgs
[  2%] Built target mavlink_c_generate
[  2%] Built target component_general_json
[ 22%] Built target uorb_msgs
[ 23%] Built target mav_msgs
[ 27%] Built target physics_msgs
[ 22%] Built target HelicopterMixer
[ 23%] Built target MultirotorMixer
[ 29%] Built target klt_feature_tracker
[ 24%] Built target parameters
[ 31%] Built target nav_msgs
[ 24%] Built target NullMixer
[ 24%] Built target SimpleMixer
[ 46%] Built target models_gen
[ 25%] Built target romfs_gen_files_target
[ 25%] Built target px4_work_queue
[ 26%] Built target mixer
[ 48%] Built target gazebo_uuv_plugin
[ 50%] Built target gazebo_lidar_plugin
[ 27%] Built target px4_platform
[ 27%] Built target lockstep_scheduler
[ 53%] Built target gazebo_user_camera_plugin
[ 55%] Built target gazebo_groundtruth_plugin
[ 28%] Built target px4_daemon
[ 29%] Built target px4_layer
[ 57%] Built target gazebo_sonar_plugin
[ 59%] Built target gazebo_camera_manager_plugin
[ 30%] Built target bezier
[ 30%] Built target hysteresis
[ 30%] Built target airspeed
[ 62%] Built target OpticalFlow
[ 30%] Built target systemlib
[ 31%] Built target cdev
[ 31%] Built target battery
[ 63%] Built target gazebo_opticalflow_mockup_plugin
[ 31%] Built target circuit_breaker
[ 31%] Built target CollisionPrevention
[ 32%] Built target controllib
[ 65%] Built target gazebo_airspeed_plugin
[ 67%] Built target gazebo_airship_dynamics_plugin
[ 32%] Built target conversion
[... 255 lines cut ...]
     [-z]        Force hardware flow control always on
     [-Z]        Force hardware flow control always off

   stop-all      Stop all instances

   stop          Stop a running instance
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>

   status        Print status for all instances
     [streams  ] Print all enabled streams

   stream        Configure the sending rate of a stream for a running instance
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>
     -s <val>    Mavlink stream to configure
     -r <val>    Rate in Hz (0 = turn off, -1 = set to default)

   boot_complete Enable sending of messages. (Must be) called as last step in startup script.
ERROR [px4] Startup script returned with return value: 65280
[100%] Built target gazebo_iris__empty
parallels@ubuntu-linux-20-04-desktop:~$ 
```

JonasVautherin, 2022-10-23:
* `uname -ar`?
* `docker --version`?

mengchaoheng, 2022-10-23:
```Console
parallels@ubuntu-linux-20-04-desktop:~/Desktop$ uname -ar
Linux ubuntu-linux-20-04-desktop 5.15.0-52-generic #58~20.04.1-Ubuntu SMP Thu Oct 13 13:10:09 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
parallels@ubuntu-linux-20-04-desktop:~/Desktop$ docker --version
Docker version 20.10.20, build 9fdeb9c
parallels@ubuntu-linux-20-04-desktop:~/Desktop$ docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.0
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
exec /root/entrypoint.sh: exec format error
```
@JonasVautherin

julianoes, 2022-10-24:
It looks like you're passing an IPv6 address rather than an IPv4, so it fails to parse that.

JonasVautherin, 2022-10-24:
Yeah something is different in this VM (looks like a parallels VM), and the script fails to detect the host IP properly:

> ERROR [mavlink] invalid partner ip '::ffff:198.18.1.14'

Also I guess it's an M1? Support for arm was added after 1.13.0, IIRC :sweat_smile: 

> WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
exec /root/entrypoint.sh: exec format error

mengchaoheng, 2022-10-24:
yes, it's m1 MacBook Pro, I running arm-based Ubuntu

JonasVautherin, 2022-10-24:
Could you run:

```
docker run --rm -it --entrypoint /bin/bash jonasvautherin/px4-gazebo-headless:1.13.1
```

And there in the terminal run:

```
ip a
ip route
```

And show the output here?

mengchaoheng, 2022-10-24:
now the ip is ok :
```Console
parallels@ubuntu-linux-20-04-desktop:~/Desktop$ docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.1
[  0%] Built target ver_gen
[  0%] Generating ../../logs
[  0%] Built target git_gazebo
[  0%] Built target parameters_xml
[  0%] Built target logs_symlink
[  0%] Built target events_header
[  0%] Built target flighttasks_generated
[  0%] Built target uorb_headers
[  0%] Built target actuators_json
[  0%] Built target drivers_board
[  0%] Built target mixer_gen_6dof
[  0%] Built target mixer_gen
[  0%] Built target output_functions_header
[  0%] Built target version
[  0%] Built target wind_estimator
[  0%] Built target world_magnetic_model
[  0%] Built target git_gps_devices
[  0%] Built target git_mavlink_v2
[  0%] Performing build step for 'sitl_gazebo'
make[5]: warning: -jN forced in submake: disabling jobserver mode.
[  0%] Built target events_json
[  0%] Built target tinybson
[  0%] Built target perf
[  2%] Built target work_queue
[  2%] Built target geo
[  3%] Built target std_msgs
[ 19%] Built target sensor_msgs
[  2%] Built target MixerBase
[  2%] Built target mavlink_c_generate
[  2%] Built target component_general_json
[ 23%] Built target physics_msgs
[ 27%] Built target mav_msgs
[ 22%] Built target uorb_msgs
[ 22%] Built target HelicopterMixer
[ 29%] Built target klt_feature_tracker
[ 31%] Built target nav_msgs
[ 23%] Built target MultirotorMixer
[ 24%] Built target parameters
[ 46%] Built target models_gen
[ 24%] Built target NullMixer
[ 24%] Built target SimpleMixer
[ 25%] Built target romfs_gen_files_target
[ 25%] Built target px4_work_queue
[ 48%] Built target gazebo_uuv_plugin
[ 26%] Built target mixer
[ 50%] Built target gazebo_lidar_plugin
[ 27%] Built target px4_platform
[ 27%] Built target lockstep_scheduler
[ 53%] Built target gazebo_user_camera_plugin
[ 55%] Built target gazebo_groundtruth_plugin
[ 28%] Built target px4_daemon
[ 29%] Built target px4_layer
[ 57%] Built target gazebo_sonar_plugin
[ 59%] Built target gazebo_camera_manager_plugin
[ 30%] Built target bezier
[ 30%] Built target hysteresis
[ 30%] Built target airspeed
[ 62%] Built target OpticalFlow
[ 63%] Built target gazebo_opticalflow_mockup_plugin
[ 31%] Built target cdev
[ 31%] Built target battery
[ 31%] Built target systemlib
[ 65%] Built target gazebo_airspeed_plugin
[ 31%] Built target circuit_breaker
[ 31%] Built target CollisionPrevention
[ 32%] Built target controllib
[ 67%] Built target gazebo_airship_dynamics_plugin
[... 239 lines cut ...]
pxh> INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative

pxh> 
pxh> 
pxh> commander takeoff
pxh> INFO  [commander] Armed by internal command	
INFO  [tone_alarm] arming warning
INFO  [navigator] Using minimum takeoff altitude: 2.50 m	
INFO  [commander] Takeoff detected	
WARN  [commander] Failsafe enabled: no RC and no datalink	
INFO  [commander] Failsafe mode activated	
INFO  [navigator] RTL HOME activated	
INFO  [navigator] RTL: landing at home position.	
INFO  [navigator] RTL: climb to 489 m (1 m above destination)	
INFO  [tone_alarm] battery warning (fast)
INFO  [commander] Failsafe mode deactivated	
INFO  [navigator] RTL: return at 489 m (1 m above destination)	
INFO  [navigator] RTL: land at destination	
INFO  [commander] Landing detected	
INFO  [commander] Disarmed by landing	
INFO  [tone_alarm] notify neutral
INFO  [logger] closed logfile, bytes written: 3787377

```

mengchaoheng, 2022-10-24:
but don't have gazebo ? @JonasVautherin

mengchaoheng, 2022-10-24:
```Console
parallels@ubuntu-linux-20-04-desktop:~/Desktop$ docker run --rm -it --entrypoint /bin/bash jonasvautherin/px4-gazebo-headless:1.13.1
root@3028512c5c40:/# 
```

JonasVautherin, 2022-10-24:
>but don't have gazebo ?

Well it's headless, you will only see the PX4 shell :sweat_smile:

mengchaoheng, 2022-10-24:
How can I use Docker like the official manual? The official image can run the sitl simulation directly, and the gazebo graphical interface can appear. like https://docs.px4.io/main/en/test_and_ci/docker.html. 

It would be great if your Docker has this feature. I think you have the ability to build one. @JonasVautherin

mengchaoheng, 2022-10-24:
> > but don't have gazebo ?
> 
> Well it's headless, you will only see the PX4 shell 😅

got it

JonasVautherin, 2022-10-24:
>now the ip is ok :

What did you change to make it work?

mengchaoheng, 2022-10-24:
> >now the ip is ok :
> 
> 
> 
> What did you change to make it work?

I don't know, maybe a new network.

JonasVautherin, 2022-10-24:
Ok, well happy to see that it works. I guess we can close it now :blush:

closed 2022-10-24 by JonasVautherin