JonasVautherin/px4-gazebo-headless #39: Running Multiple Containers

OliverHeilmann, 2023-08-11:
Hello! I am currently exploring running multiple simulators in parallel for some CICD pipeline testing. I have used the [Multi-Vehicle-Simulation-With-Gazebo](https://docs.px4.io/v1.12/en/simulation/multi_vehicle_simulation_gazebo.html#multi-vehicle-simulation-with-gazebo) which has been useful in checking that my agents can connect to multiple assets/ vehicles. *Note: [this](https://discuss.px4.io/t/multi-vehicle-simulation-where-is-gazebo-sitl-multiple-run-sh/29406) helped me find the correct shell script.*

My issue with this is that I cannot run the simulation in a headless environment i.e. on a pipeline test. I was thinking about containerising the multi-vehicle simulation. Gut feel is that three individual docker images may actually be the simplest option but I don't know how to map the `UDP:14540` port to other ports as you aren't using the traditional docker `-p` command for it (noting that this approach would have three instances of Gazebo running which isn't great either).

So my question is two fold:
1) Do you know how to run a mutli-vehicle headless simulation in a docker container? I can handle the mapping of ports 14540, 14541... 1454n using normal docker practices.
2) If this is not possible, how can I modify this docker image to allow for remapping port 14540 to another one as done with the normal docker approach?

JonasVautherin, 2023-08-11:
For the first question, I don't think anything prevents you from running a multi-vehicle headless simulation with some changes to this image. I haven't looked into it though, it would require a couple changes.

For the second question, the ports are defined here: https://github.com/JonasVautherin/px4-gazebo-headless/blob/master/edit_rcS.bash#L59C35-L59C58. So you could probably change `\$udp_offboard_port_local` for the port you want.

OliverHeilmann, 2023-08-12:
Thanks for the prompt response! I will look into modifying this image/ dockerfile to support multi-vehicle simulation. I think this is a more elegant solution over running multiple containers, with each running a simulator. Perhaps I will open up a pull request if I arrive to an elegant solution...

Regarding the `\$udp_offboard_port_local`, I'll give this a try but I am happy to close this issue now. Thanks!

OliverHeilmann, 2023-08-14:
Hello again! 

Thought I'd reopen this issue instead of making a new one so the discussion stays on one thread... Below are the changes I have made in my attempt to get the multi-vehicle simulation working in a docker container. I used this repo as a baseline and made changes to a couple of the scripts.

## [entrypoint.sh](https://github.com/JonasVautherin/px4-gazebo-headless/blob/ce2e0f55b21118533f973d558e2eb9dc22f05ccb/entrypoint.sh#L67)
Changing the `HEADLESS=1 make ...` line to the `sitl_multiple_run.sh` made sense, the issue here is that the shell script uses the pre-built binaries instead of a make command directly.
```diff
+ ./Tools/simulation/gazebo-classic/sitl_multiple_run.sh -s "${vehicle}:3" -w "${world}"
-  HEADLESS=1 make px4_sitl gazebo_${vehicle}__${world}
```

## [sitl_multiple_run.sh](https://github.com/PX4/PX4-Autopilot/blob/4fce159a2ab490410e730d492d4135f1beca2db4/Tools/simulation/sitl_multiple_run.sh#L31)
Add headless option to the px4 referenced binaries to ensure it uses the headless mode instead.
```diff
+ HEADLESS=1 $build_path/bin/px4 -i $n -d "$build_path/etc" >out.log 2>err.log &
- $build_path/bin/px4 -i $n -d "$build_path/etc" >out.log 2>err.log &
```

## Issue with REST Request
When I run the below console command, being sure to map the other UDP ports which will be used for other vehicles (following the [documentation](https://docs.px4.io/v1.12/en/simulation/multi_vehicle_simulation_gazebo.html)), I get part of the way there.
```console
oliver@ubuntu:~/<DIR>$ docker run --name sitl -p 14541:14541 -p 14542:14542 --rm -it px4-sitl-headless:latest
```

From the screenshot below, you can see that all three vehicles are created in the docker container and the headless mode is in fact working. The two errors are relating to an Audio device (which perhaps is not an issue since audio is then disabled) and then the REST request. The programme then hangs on the `libcurl` print and doesn't do anything else.

I have however tested the connections to the vehicles in the docker containers and it seems to be working which is great however. Any idea on these issues? I suspect if you follow those two changes, you should be able to replicate my issue.

[image]

Thanks again for your help!

pushkarprasad007, 2025-09-13:
@JonasVautherin - as mentioned in [link](https://github.com/mavlink/MAVSDK-Python/issues/800), for me - the requirement is to simulate two vehicle (different behavior - and also type - plane & vtol) at same time, using mavsdk-python code.

I'm inclined to believe if we build the docker image - with the separate ports - should work. Then why do you say this:

> What you are trying to do above won't work, though.

Can you please elaborate?

Also, I'm not super confident on how to make this work (possible to help - pretty please):

>  In your case, maybe you would want to look into simulating two drones in the same px4-gazebo-headless instance (maybe using mavlink-router to separate the streams outside of the container if necessary).

IvanC-Gambit, 2025-09-15:
> Has anyone been using that in the meantime?

I was able to modify the following repo to get multiple vehicles simulated. Required some modifications. Interested in PR upstream to your repo?

pushkarprasad007, 2025-09-15:
Yes please @IvanC-Gambit !!

JonasVautherin, 2025-09-15:
@IvanC-Gambit: sure, open for PRs!