JonasVautherin/px4-gazebo-headless #41: Euler Angles are NaN

poli0048, 2023-10-27:
I am using MAVSDK to connect to this simulator over UDP and I am finding that when I use attitude_euler() from the Telemetry plugin all of the fields are always NaN. Is this because orientation data isn't being provided by the sim, or is this an indicator that I am doing something wrong? Thanks.

JonasVautherin, 2023-10-28:
This container is running upstream PX4, I think it should send the same messages as upstream PX4. Did you try checking the message with something else (e.g. QGC) to verify that it is not coming from MAVSDK? And did you try using another autopilot (e.g. a pixhawk running PX4) to verify that it is not coming from this simulator? Would be helpful :blush:

poli0048, 2023-10-30:
Hi Jonas, Thanks for the quick reply. This is helpful. It seems QGC is getting good attitude data somehow so it looks like the Sim is not the issue here. I haven't tried with a physical system yet (still waiting on a few parts to arrive). I have a better idea of where to dig now. Best.

closed 2023-10-30 by poli0048