JonasVautherin/px4-gazebo-headless #48: Unable to connect to QGC and MavSDK

KarthiAru, 2024-06-12:
Hi,

I'm unable to connect to the PX4 SITL using both QGC and MavSDK.

This is the MavSDK python script. It is stuck in the `drone.connect` step.

```
import asyncio
from mavsdk import System

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        print(state)  # Print the connection state for debugging
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

asyncio.run(run())
```

I'm sharing the logs from the docker container

```
(base)  ~  docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3
Scanning dependencies of target gazebo_iris__empty
WARNING gazebo_iris__empty target DEPRECATED, please use gazebo-classic_iris__empty
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  0%] Built target parameters_xml
[  0%] Built target ver_gen
[  0%] Built target events_header
[  0%] Built target uorb_headers
[  0%] Built target flighttasks_generated
[  1%] Built target events_json
[  1%] Built target actuators_json
[  1%] Built target component_general_json
[  1%] Built target romfs_gen_files_target
[ 22%] Built target uorb_msgs
[ 22%] Built target geo
[ 22%] Built target perf
[ 22%] Built target tinybson
[ 24%] Built target work_queue
[ 24%] Built target drivers_board
[ 24%] Built target parameters
[ 25%] Built target px4_work_queue
[ 27%] Built target px4_platform
[ 27%] Built target lockstep_scheduler
[ 27%] Built target px4_daemon
[ 28%] Built target px4_layer
[ 28%] Built target adsb
[ 29%] Built target airspeed
[ 30%] Built target bezier
[ 30%] Built target hysteresis
[ 30%] Built target avoidance
[ 30%] Built target battery
[ 30%] Built target cdev
[ 30%] Built target systemlib
[ 31%] Built target uORB
[ 31%] Built target lib__cdev__test__cdev_test
[ 31%] Built target circuit_breaker
[ 31%] Built target CollisionPrevention
[ 34%] Built target controllib
[ 34%] Built target lib__controllib__controllib_test
[ 34%] Built target conversion
[ 34%] Built target drivers_accelerometer
[ 35%] Built target drivers_gyroscope
[ 35%] Built target drivers_magnetometer
[... 557 lines cut ...]
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 39: [: Illegal number: 
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO  [lockstep_scheduler] setting initial absolute time to 1012000 us
INFO  [commander] LED: open /dev/led0 failed (22)
ERROR [param] Parameter UXRCE_DDS_DOM_ID not found.
etc/init.d-posix/rcS: 292: etc/init.d-posix/rcS: uxrce_dds_client: not found
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-06-12/17_30_47.ulg	
INFO  [logger] Opened full log file: ./log/2024-06-12/17_30_47.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] notify negative
INFO  [commander] Ready for takeoff!
```

KarthiAru, 2024-06-12:
It is working with both the QGC and MAVSDK python scripts using [gazebo-jmavsim](https://docs.px4.io/v1.12/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets).

JonasVautherin, 2024-06-12:
On which system are you running? Can you try the previous version, just to make sure:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.2
```

KarthiAru, 2024-06-13:
It is working with both the QGC and MAVSDK python scripts using [gazebo-jmavsim](https://docs.px4.io/v1.12/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets).

> On which system are you running? Can you try the previous version, just to make sure:
> 
> ```
> docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.2
> ```

I'm running on Ubuntu 22.04. 1.14.2 wasn't working either.
I suspected some issue with my docker setup and [troubleshooted](https://askubuntu.com/questions/1423293/ubuntu-22-04-docker-containers-not-accessible-from-outside) it. After setting the FORWARD chain policy to ACCEPT, it is working now for both 1.14.2 and 1.14.3.

closed 2024-06-13 by KarthiAru