JonasVautherin/px4-gazebo-headless #49: Windows 11 - MAVLink traffic for UDP port not reaching QGC and MAVSDK

KarthiAru, 2024-06-30:
Hi,

The MAVLink traffic for UDP port 14550 (and 14540) does not seem to arrive in QGroundControl (and MAVSDK) on Windows 11 OS. I've installed Docker Desktop on Windows. I've also added two rules for incoming and outgoing connections for the UDP port in the Windows Firewall.

I tried running the following commands, but no luck.
```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 10.0.2.2 10.0.2.2
docker run --rm --network host -it jonasvautherin/px4-gazebo-headless:1.14.3 127.0.0.1 127.0.0.1
```

Logs:
```
PS C:\Users\DELL> docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3
Scanning dependencies of target gazebo_iris__empty
WARNING gazebo_iris__empty target DEPRECATED, please use gazebo-classic_iris__empty
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  0%] Built target parameters_xml
[  0%] Built target ver_gen
[  0%] Built target events_header
[  0%] Built target uorb_headers
[  0%] Built target flighttasks_generated
[  1%] Built target events_json
[  1%] Built target actuators_json
[  1%] Built target component_general_json
[  1%] Built target romfs_gen_files_target
[ 22%] Built target uorb_msgs
[ 22%] Built target geo
[ 22%] Built target perf
[ 22%] Built target tinybson
[ 24%] Built target work_queue
[ 24%] Built target drivers_board
[ 24%] Built target parameters
[ 25%] Built target px4_work_queue
[ 27%] Built target px4_platform
[ 27%] Built target lockstep_scheduler
[ 27%] Built target px4_daemon
[ 28%] Built target px4_layer
[ 28%] Built target adsb
[ 29%] Built target airspeed
[ 30%] Built target bezier
[ 30%] Built target hysteresis
[ 30%] Built target avoidance
[ 30%] Built target battery
[ 30%] Built target cdev
[ 30%] Built target systemlib
[ 31%] Built target uORB
[ 31%] Built target lib__cdev__test__cdev_test
[ 31%] Built target circuit_breaker
[ 31%] Built target CollisionPrevention
[ 34%] Built target controllib
[ 34%] Built target lib__controllib__controllib_test
[ 34%] Built target conversion
[ 34%] Built target drivers_accelerometer
[ 35%] Built target drivers_gyroscope
[ 35%] Built target drivers_magnetometer
[ 35%] Built target l1
[ 35%] Built target mathlib
[ 35%] Built target output_functions_header
[ 35%] Built target mixer_module
[ 35%] Built target motion_planning
[ 35%] Built target npfg
[ 35%] Built target pid
[ 35%] Built target RateControl
[ 35%] Built target rc
[ 35%] Built target lib__rc__rc_tests
[ 36%] Built target ringbuffer
[ 37%] Built target sensor_calibration
[ 37%] Built target SlewRate
[ 38%] Built target SystemIdentification
[... 231 lines cut ...]
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 39: [: Illegal number:
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO  [lockstep_scheduler] setting initial absolute time to 576000 us
INFO  [commander] LED: open /dev/led0 failed (22)
ERROR [param] Parameter UXRCE_DDS_DOM_ID not found.
etc/init.d-posix/rcS: 292: etc/init.d-posix/rcS: uxrce_dds_client: not found
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-06-30/07_49_58.ulg
INFO  [logger] Opened full log file: ./log/2024-06-30/07_49_58.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] notify negative
INFO  [commander] Ready for takeoff!
```

KarthiAru, 2024-06-30:
> [These](https://github.com/JonasVautherin/px4-gazebo-headless/blob/master/edit_rcS.bash#L12-L24) are the tricks I was doing for the Docker VMs (Windows/macOS), but I haven't tried them in years (I work on Linux). Maybe this has changed.
> 
> Let me know if you find a way to fix this on Windows 11. A first thing you could try would be to run a simple Linux image in a docker container, check the network setup and see how you can e.g. ping the Windows host.
> 
> Also, is Docker still running in a VM on Windows? Or do they somehow leverage WSL?

Docker Desktop v4.31.1 (latest) uses WSL2 based engine by default. 
I spun up a Ubuntu container and was able to ping the host from the container. Also I was able to send and receive UDP message from the container to the host.

Sending a UDP packet from Docker container
```
root@docker-desktop:/# echo "Hello QGC" | nc -u 192.168.0.138 14550
```

Listening for UDP packet from host
```
PS C:\Users\karth\Documents> python test.py
received message: Hello QGC
 from ('192.168.0.138', 63447)
 ```

test.py
```
import socket

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.bind(('0.0.0.0', 14550))

while True:
    data, addr = sock.recvfrom(1024)
    print("received message:", data.decode(), "from", addr)
```

JonasVautherin, 2024-07-01:
Does this mean that 192.168.0.138 is the IP of the host from WSL? In that case did you try the following?

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 192.168.0.138 192.168.0.138
```

Miguelmcanabal, 2024-10-10:
Try downgrading to Docker Desktop 4.30.0, it worked for me

Pjort, 2025-04-30:
I can confirm the problem is still there with version 4.40.0 and 4.41.0.

But downgrading to 4.30.0 also worked for me.

JonasVautherin, 2025-04-30:
From the container, can you show the output of those two commands?

```
getent hosts host.docker.internal
```

and

```
getent ahostsv4 host.docker.internal
```

Pjort, 2025-05-01:
> From the container, can you show the output of those two commands?
> 
> ```
> getent hosts host.docker.internal
> ```
> 
> and
> 
> ```
> getent ahostsv4 host.docker.internal
> ```

root@54287650ac02:/# getent hosts host.docker.internal
fdc4:f303:9324::254 host.docker.internal
root@54287650ac02:/# getent ahostsv4 host.docker.internal
192.168.65.254  STREAM host.docker.internal
192.168.65.254  DGRAM
192.168.65.254  RAW

JonasVautherin, 2025-05-01:
And would this work, by any chance?

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 192.168.65.254 192.168.65.254
```

Pjort, 2025-05-01:
> And would this work, by any chance?
> 
> ```
> docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 192.168.65.254 192.168.65.254
> ```

Yes, it does infact work.

closed 2025-05-02 by JonasVautherin

JonasVautherin, 2025-05-03:
@KarthiAru: Would you mind trying with the latest image and let us know if it works? With this:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:latest
```

KarthiAru, 2025-05-04:
@JonasVautherin It works. Thanks!

PR #55 (2025-05-02): Switch to ahostsv4
Supposedly, this should work for macOS and Win11?

Resolves #49, Resolves #54