JonasVautherin/px4-gazebo-headless #50: Can't connect to UDP connection exposed by Docker

Swabhan, 2024-09-18:
Running the command ```docker run --rm -it -p 14540:14540/udp jonasvautherin/px4-gazebo-headless:1.14.3 ```, I am able to successfully open Gazebo in the terminal. My goal is to have the simulation send data via a UDP port on 14540 so I can be able to receive data with MavSDK on C++. While I understand the following problem is most likely a problem not connected with your docker image, I was wondering if you have found a way to work around this issue. 

This image is of the running Simulation within the container
[image]

This image is of a bind error caused by connecting with code after the image
[image]

```
int connect_drone(std::string connection_url = "udp://:14540", bool subscribeToData = true){
    //Connects to UDP
    std::cout << "Listening on " << connection_url << std::endl;

    ConnectionResult connection_result = mavsdkConnect.add_any_connection(connection_url);

    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << std::endl;
        return -1;
    }

    //Waits for connection
    std::cout << "Waiting for drone to connect..." << std::endl;
    std::promise<std::shared_ptr<System>> prom;
    std::future<std::shared_ptr<System>> fut = prom.get_future();

```

The below image is showing the UDP connection is made 

[image]

JonasVautherin, 2024-09-18:
What do you expect `-p 14540:14540/udp` to do? It will redirect 14540 from your host to the docker container, which I think doesn't do anything about it.

By default px4-gazebo-headless sends MAVLink to the host on 14540 (so `udp://:14540` should work on the host).

On which OS are you running?

Swabhan, 2024-09-18:
> What do you expect `-p 14540:14540/udp` to do? It will redirect 14540 from your host to the docker container, which I think doesn't do anything about it.
> 
> By default px4-gazebo-headless sends MAVLink to the host on 14540 (so `udp://:14540` should work on the host).
> 
> On which OS are you running?

I am running MacOS (M3 Arm64), but I am unable to connect through my C++ program with MavSDK
[image]
It is stuck on this within the command line

JonasVautherin, 2024-09-18:
> I am running MacOS (M3 Arm64)

Hmm I have never tried this setup. Doesn't Docker run in a VM on macOS? px4-gazebo-headless used to work on macOS, but that was years ago.

A question would be this: when you are inside a docker container, what is the IP of the host? Then you could try to run:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 <ip_of_the_host>
```

Swabhan, 2024-09-18:
> Hmm I have never tried this setup. Doesn't Docker run in a VM on macOS? px4-gazebo-headless used to work on macOS, but that was years ago.
>
> A question would be this: when you are inside a docker [...]

Solved!

```docker run --rm alpine sh -c "getent ahostsv4 host.docker.internal | awk '{print \$1}' | uniq"``` 

The above command is to recieve the IP Address

```docker run --rm -it jonasvautherin/px4-gazebo-headless:1.14.3 <ip_of_the_host>```

The above command runs it

closed 2024-09-18 by Swabhan

JonasVautherin, 2024-09-19:
Nice!

My script runs [`getent hosts host.docker.internal | awk '{ print $1 }'`](https://github.com/JonasVautherin/px4-gazebo-headless/blob/master/edit_rcS.bash#L23). Could you tell me what it outputs for you?

Would be nice if we could fix my script so that it just works :+1: