JonasVautherin/px4-gazebo-headless #52: Ubuntu 22.04

garrettseepersad, 2024-11-12:
If I wanted to modify the dockerfile to docker 22.04, what additional changes are needed?

julianoes, 2024-11-12:
What are you trying to do? Which dockerfile?

JonasVautherin, 2024-11-13:
Seems like [this is enough](https://github.com/JonasVautherin/px4-gazebo-headless/pull/53). @garrettseepersad would you be willing to try it?

garrettseepersad, 2024-11-13:
Thanks @JonasVautherin, I'll give it a test run. Appreciate you taking a first stab at it.

garrettseepersad, 2024-11-20:
Hello @JonasVautherin , after reading your thread [here](https://github.com/JonasVautherin/px4-gazebo-headless/pull/53), looks like it's better to focus on Ubuntu 24.04

JonasVautherin, 2025-01-15:
I started looking into it [here](https://github.com/JonasVautherin/px4-gazebo-headless/tree/add-gz-harmonic)

closed 2025-01-15 by JonasVautherin

JonasVautherin, 2025-09-19:
FYI: `master` is now running on Ubuntu 24.04 with gazebo-harmonic.