JonasVautherin/px4-gazebo-headless #54: "docker run" generated "invalid partner ip 'fdc4:f303:9324::254' " error

ytl001, 2025-01-10:
This looks similar to https://github.com/JonasVautherin/px4-gazebo-headless/issues/34, but that error seemed to have disappeared mysteriously without any specific action.

px4 did come up at the end, but a running instance of QGroundControl (on the host machine) did not connect.  I should be running QGroundControl outside docker, correct?

MacOS Ventura 13.7.2
docker:
Desktop 4.37.1
Engine 27.4.0
Compose v2.31.0-desktop.2
Kubernetes: v1.30.5
Credential Helper: v0.8.2
-----------------------------------------------------------------------------------

Yings-iMac:~ ytl$ docker run --rm -it jonasvautherin/px4-gazebo-headless
Scanning dependencies of target gazebo_iris__empty
WARNING gazebo_iris__empty target DEPRECATED, please use gazebo-classic_iris__empty
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  0%] Built target parameters_xml
[  0%] Built target ver_gen
[  1%] Built target events_header
[  1%] Built target uorb_headers
[  1%] Built target flighttasks_generated
[  1%] Built target events_json
[  1%] Built target actuators_json
[  1%] Built target component_general_json
[  1%] Built target romfs_gen_files_target
[ 22%] Built target uorb_msgs
[ 22%] Built target geo
[ 22%] Built target perf
[ 22%] Built target tinybson
[ 25%] Built target work_queue
[ 26%] Built target drivers_board
[ 26%] Built target parameters
[ 26%] Built target px4_work_queue
[ 28%] Built target px4_platform
[ 28%] Built target lockstep_scheduler
[ 28%] Built target px4_daemon
[ 29%] Built target px4_layer
[ 29%] Built target adsb
[ 30%] Built target airspeed
[ 30%] Built target atmosphere
[ 31%] Built target bezier
[ 31%] Built target hysteresis
[ 31%] Built target avoidance
[ 31%] Built target battery
[ 31%] Built target cdev
[ 31%] Built target heatshrink
[ 32%] Built target systemlib
[ 32%] Built target uORB
[ 32%] Built target events
[ 32%] Built target lib__cdev__test__cdev_test
[ 32%] Built target circuit_breaker
[ 33%] Built target CollisionPrevention
[ 34%] Built target controllib
[ 34%] Built target lib__controllib__controllib_test
[ 35%] Built target conversion
[ 35%] Built target dataman_client
[ 35%] Built target drivers_accelerometer
[ 35%] Built target drivers_gyroscope
[ 35%] Built target drivers_magnetometer
[ 35%] Built target l1
[ 35%] Built target mathlib
[ 36%] Built target output_functions_header
[ 38%] Built target mixer_module
[ 38%] Built target motion_planning
[ 38%] Built target npfg
[ 38%] Built target performance_model
[ 38%] Built target pid
[ 38%] Built target RateControl
[ 39%] Built target rc
[... 350 lines cut ...]
                 values: <file:dev>

   status        Print status for all instances
     [streams  ] Print all enabled streams

   stream        Configure the sending rate of a stream for a running instance
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>
     -s <val>    Mavlink stream to configure
     -r <val>    Rate in Hz (0 = turn off, -1 = set to default)

   boot_complete Enable sending of messages. (Must be) called as last step in startup script.
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-01-10/00_39_32.ulg	
INFO  [logger] Opened full log file: ./log/2025-01-10/00_39_32.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [commander] Ready for takeoff!
WARN  [health_and_arming_checks] Preflight Fail: height estimate not stable

pxh>

ytl001, 2025-01-13:
> fdc4:f303:9324::254 looks like an IPv6, right? It seems like my script
> detects an IPv6 for the host and tries to pass it to MAVLink, which doesn't
> support it 🤔.
>
> Could you try to get into the container with something like:
>
> docker run --rm -it --entrypoint /bin/bash jonasvautherin/px4-gazebo-headless
>
> And from there show me the output of:
>
>  ip route
>
> and
>
> ip route | awk '/default/ { print $3 }'
>
> I expect the latter to output fdc4:f303:9324::254.

Thank you for your prompt response.  I don't think I'm using IPv6, though
it does look like an IPv6 address with some missing numbers.  Here's the
output you requested.

Yings-iMac:~ ytl$ docker run --rm -it --entrypoint /bin/bash
jonasvautherin/px4-gazebo-headless
***@***.***:/# ip route
default via 172.17.0.1 dev eth0
172.17.0.0/16 dev eth0 proto kernel scope link src 172.17.0.2
***@***.***:/# ip route | awk '/default/ { print $3 }'
172.17.0.1
***@***.***:/#

JonasVautherin, 2025-01-13:
And can you try running:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless 172.17.0.1 172.17.0.1
```

Does that work better?

ytl001, 2025-01-13:
Yes, the ip error went away when px4 came up.

QGroundControl is still not connecting though.  Do I need to set something
in QGroundControl?  I started it before running the docker command and it
kept showing "Disconnected".  Or did I misunderstand how QGroundControl
works with the simulator?

Thank you very much for your help!

JonasVautherin, 2025-01-14:
Is QGC running on the host? It should receive MAVLink on 14550 :thinking:

ytl001, 2025-01-15:
QGC is running on the host.  But it was spawned in a different process than
the shell performing docker run.  This should be ok, right?

I'm going to be out of town for the next couple of weeks.  I will work on
it more when I get back.

Thank you so much.

JonasVautherin, 2025-01-15:
> But it was spawned in a different process than the shell performing docker run.  This should be ok, right?

Yes that is fine.

One thing is that I don't know if it still works on macOS. I made it work 7 years ago and haven't tried it since, chances are that something may have changed with docker :see_no_evil:.

julianoes, 2025-02-04:
I saw the same problem with docker on macOS, and adding the IPs fixed it as suggested in https://github.com/JonasVautherin/px4-gazebo-headless/issues/54#issuecomment-2588396847. I wonder why this happens.

JonasVautherin, 2025-02-05:
Would be interesting to know where this number comes from. It's supposed to come from the `ip route` command, but apparently it did not appear there 🤔

julianoes, 2025-02-05:
Must be this line, right?

https://github.com/JonasVautherin/px4-gazebo-headless/blob/2b20facd9002fcdd9b3df702b38b81b426a47199/edit_rcS.bash#L23

julianoes, 2025-02-05:
I will ask what this is on that machine.

JonasVautherin, 2025-02-05:
Oh maybe it is, yes. I was looking at this one, which seems to work for @ytl001:

https://github.com/JonasVautherin/px4-gazebo-headless/blob/2b20facd9002fcdd9b3df702b38b81b426a47199/edit_rcS.bash#L27

Maybe the `is_docker_vm` logic is obsolete for macOS and it should just use the normal `ip route` way now?

julianoes, 2025-02-06:
My datapoint is that this line gives on macOS docker:

```
getent hosts host.docker.internal  this returned
fdc4:f303:9324::254 host.docker.internal
```

ytl001, 2025-02-11:
Jonas,

I notice that the px4 (running inside docker) says:
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1
to enable network)

and indeed the command "mavlink status" returns:
instance #0:
mavlink chan: #0
type: GENERIC LINK OR RADIO
flow control: OFF
rates:
 tx: 17059.8 B/s
 txerr: 0.0 B/s
 tx rate mult: 1.000
 tx rate max: 4000000 B/s
 rx: 0.0 B/s
 rx loss: 0.0%
FTP enabled: YES, TX enabled: YES
mode: Normal
Forwarding: On
MAVLink version: 2
transport protocol: UDP (18570, remote port: 14550)
Broadcast enabled: NO
partner IP: 172.17.0.1

instance #1:
mavlink chan: #1
type: GENERIC LINK OR RADIO
flow control: OFF
rates:
 tx: 24971.1 B/s
 txerr: 0.0 B/s
 tx rate mult: 1.000
 tx rate max: 4000000 B/s
 rx: 0.0 B/s
 rx loss: 0.0%
FTP enabled: YES, TX enabled: YES
mode: Onboard
Forwarding: On
MAVLink version: 2
transport protocol: UDP (14580, remote port: 14540)
Broadcast enabled: NO
partner IP: 172.17.0.1

But after I issue the command "param set MAV_0_BROADCAST 1", the status
still shows the same "Broadcast enabled: NO" as before.  I found a solution
in: https://discuss.px4.io/t/cannot-enable-mav-i-broadcast/22408, but I
think you will have to do this in your docker build code.

I suspect this might be the problem, since px4 seems to be able to see
qgroundcontrol but qgroundcontrol cannot see px4 inside the container when
I set up a comm link in qgroundcontrol to point to the local port assigned
by px4 (172.17.0.2:18570, where 172.17.0.2 is the ip address of the
container).  When I set the comm link to be 172.17.0.1:14550 (where
172.17.0.1 is the ip address of the host), px4 and qgroundcontrol cannot
see each other.

Let me know what you think.

Thanks,
Ying

JonasVautherin, 2025-02-12:
I think there is some confusion here. Your MAVLink output says:

```
transport protocol: UDP (18570, remote port: 14550)
partner IP: 172.17.0.1
...
transport protocol: UDP (14580, remote port: 14540)
partner IP: 172.17.0.1
```

Which means that MAVLink is sending to `172.17.0.1:14550` and `172.17.0.1:14540`. You should not have to care about the broadcast at all. We don't want to broadcast, we just want to send it to the host (172.17.0.1).

In QGC, you don't want to set a comm link sending to e.g. `172.17.0.2:18570`. You just want to listen to 14550/udp, which should be the default in QGC.

ytl001, 2025-02-12:
Thank you for the explanation.

Running QGC in default and turning on LinkManagerVerboseLog, it seems that
it is only listening to a bluetooth incoming port:

[image: Screenshot 2025-02-12 at 10.34.34 AM.png]

Or I am not understanding the log correctly?

Strangely enough, when I use a comm link in QGC to point to 172.17.0.2:18570,
I got a ground station heartbeat in px4, even though QGC still shows
"Disconnected":
instance #0:
GCS heartbeat valid
mavlink chan: #0
type: GENERIC LINK OR RADIO
flow control: OFF
rates:
 tx: 17063.7 B/s
 txerr: 0.0 B/s
 tx rate mult: 1.000
 tx rate max: 4000000 B/s
 rx: 16.9 B/s
 rx loss: 0.0%
Received Messages:
 sysid:255, compid:190, Total: 11 (lost: 0)
   msgid:    0, Rate:  1.0 Hz, last 0.37s ago
FTP enabled: YES, TX enabled: YES
mode: Normal
Forwarding: On
MAVLink version: 2
transport protocol: UDP (18570, remote port: 14550)
Broadcast enabled: NO
partner IP: 172.17.0.1

instance #1:
mavlink chan: #1
type: GENERIC LINK OR RADIO
flow control: OFF
rates:
 tx: 24984.1 B/s
 txerr: 0.0 B/s
 tx rate mult: 1.000
 tx rate max: 4000000 B/s
 rx: 0.0 B/s
 rx loss: 0.0%
FTP enabled: YES, TX enabled: YES
mode: Onboard
Forwarding: On
MAVLink version: 2
transport protocol: UDP (14580, remote port: 14540)
Broadcast enabled: NO
partner IP: 172.17.0.1

BTW, I am running docker-mac-net-connect to make sure that the host can
ping the docker container.  Turning docker-mac-net-connect off doesn't seem
to make any difference though.

Thank you again for your patience with a newbie.

JonasVautherin, 2025-02-12:
> Or I am not understanding the log correctly?

No image was uploaded. Did you try adding a UDP comm link that listens on port 14550? Don't enter an IP or anything in the second field, just enter the port. Also can you show a screenshot of your comm link setup?

ytl001, 2025-02-12:
I uploaded the screenshot on the QGC console.  Sorry about this; I use gmail to reply to the thread, but evidently it is not completely compatible.

This is the comm link which gives a valid heartbeat in px4 but still does not get QGC connected:

[image]

I also tried setting a comm link for port 14550:

[image]

After I click connect using this link, nothing happens - no heartbeat detected in px4 under "mavlink status", QGC remains disconnected, QGC console shows only a bluetooth port being listened to (as shown in the screenshot in https://github.com/JonasVautherin/px4-gazebo-headless/issues/54#issuecomment-2654556369).

JonasVautherin, 2025-02-13:
Hmm I don't understand then, 14550 should do it.

ytl001, 2025-04-17:
I solved my issue.  The line of code that generated the incorrect ip address fdc4:f303:9324::254 for the host is:
getent hosts host.docker.internal
which is in edit_rcS.bash.

I used
getent ahostsv4 host.docker.internal | awk '{print $1}'
to get the correct ip address of the host, which is 192.168.65.254 in my case.  Then everything works using:
docker run --rm -it yingtatleung/px4broadcast-gazebo-headless 192.168.65.254 192.168.65.254

closed 2025-04-17 by ytl001

julianoes, 2025-04-29:
@ytl001 can we fix this for the next person?

JonasVautherin, 2025-05-03:
@ytl001: Would you mind trying with the latest image and let us know if it works? With this:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:latest
```

ytl001, 2025-05-06:
Looks like we got the same error inside px4 as before: 
ERROR [mavlink] invalid partner ip 'fdc4:f303:9324::254'

Yings-iMac:Exnada ytl$ docker run --rm -it jonasvautherin/px4-gazebo-headless:latest
Scanning dependencies of target gazebo_iris__empty
WARNING gazebo_iris__empty target DEPRECATED, please use gazebo-classic_iris__empty
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  1%] Built target parameters_xml
[  1%] Built target ver_gen
[  2%] Built target events_header
[  2%] Built target uorb_headers
[  2%] Built target flighttasks_generated
[  2%] Built target events_json
[  2%] Built target actuators_json
[  2%] Built target component_general_json
[  2%] Built target romfs_gen_files_target
[ 24%] Built target uorb_msgs
[ 24%] Built target geo
[ 24%] Built target perf
[ 24%] Built target tinybson
[ 26%] Built target work_queue
[ 26%] Built target drivers_board
[ 27%] Built target parameters
[ 28%] Built target px4_work_queue
[ 29%] Built target px4_platform
[ 29%] Built target lockstep_scheduler
[ 30%] Built target px4_daemon
[ 31%] Built target px4_layer
[ 31%] Built target adsb
[ 31%] Built target airspeed
[ 32%] Built target atmosphere
[ 32%] Built target bezier
[ 32%] Built target hysteresis
[ 32%] Built target avoidance
[ 32%] Built target battery
[ 32%] Built target cdev
[ 32%] Built target heatshrink
[ 32%] Built target systemlib
[ 34%] Built target uORB
[ 34%] Built target events
[ 34%] Built target lib__cdev__test__cdev_test
[ 34%] Built target circuit_breaker
[ 34%] Built target mathlib
[ 34%] Built target CollisionPrevention
[ 34%] Built target ControlAllocation
[ 34%] Built target PID
[ 34%] Built target ActuatorEffectiveness
[ 35%] Built target controllib
[ 35%] Built target lib__controllib__controllib_test
[ 35%] Built target conversion
[ 35%] Built target dataman_client
[ 35%] Built target drivers_accelerometer
[ 35%] Built target drivers_gyroscope
[ 35%] Built target drivers_magnetometer
[ 35%] Built target l1
[ 35%] Built target lat_lon_alt
[ 35%] Built target output_functions_header
[ 36%] Built target mixer_module
[ 37%] Built target motion_planning
[ 37%] Built target npfg
[ 37%] Built target performance_model
[ 37%] Built target pure_pursuit
[ 37%] Built target RateControl
[ 38%] Built target rc
[ 39%] Built target lib__rc__rc_tests
[ 39%] Built target ringbuffer
[ 39%] Built target rover_control
[ 39%] Built target rtl_time_estimator
[ 39%] Built target sensor_calibration
[ 40%] Built target SlewRate
[... 352 lines cut ...]
   stop          Stop a running instance
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>

   status        Print status for all instances
     [streams  ] Print all enabled streams

   stream        Configure the sending rate of a stream for a running instance
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>
     -s <val>    Mavlink stream to configure
     -r <val>    Rate in Hz (0 = turn off, -1 = set to default)

   boot_complete Enable sending of messages. (Must be) called as last step in startup script.
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-05-06/05_56_20.ulg	
INFO  [logger] Opened full log file: ./log/2025-05-06/05_56_20.ulg
INFO  [px4] Startup script returned successfully
pxh> INFO  [commander] Ready for takeoff!

JonasVautherin, 2025-05-06:
@ytl001: feels like it's using a cached docker image on your system. Can you delete it and run the command again? Make sure it downloads the image 👍.

ytl001, 2025-05-06:
Indeed I was using an old copy; sorry about that.  I thought the "latest" tag would do it, but not really.

I'm happy to report that all is now working well.  Just the plain docker run command would do it:
docker run --rm -it jonasvautherin/px4-gazebo-headless:latest

QGC will connect immediately after coming up.

JonasVautherin, 2025-05-07:
Amazing! Thank you for testing!

PR #55 (2025-05-02): Switch to ahostsv4
Supposedly, this should work for macOS and Win11?

Resolves #49, Resolves #54