JonasVautherin/px4-gazebo-headless #58: INSTRUCTIONS for QGC to auto-connect to this SITL NOT ACCURATE - successful build/run on Windows 10 Pro but UNSUCCESSFUL connection

markvogt, 2025-05-14:
Greetings Jonas, 

I was really disappointed. 
I spent several evenings repeatedly running the docker "broadcast" commands over & over, each one succesful. 
Yet when I started QGC... NOTHING HAPPENED. 
QGC launched, quite alright. But all it showed was the QGC on the map - NO DRONE/SITL. 

ARE your instructions COMPLETE & ACCURATE? 

This is really really frustrating. 

My system is this: 
- Laptop: Windows 10 Pro 16 GB of RAM (plenty to run your Docker container); 
- Docker Desktop: latest version 28.1.1 (Docker engine)
- Network: home WiFi

MY BIGGEST QUESTIONS: 
1. WHAT AM I DOING WRONG ??????? 
2. HOW DO I FIX IT ???? 
3. IS ANYONE ELSE actually SUCCESSFUL with your claim "just run this" ?? 

Hope to hear from you, as I'M running (Windows 10 Pro) what is arguably the MOST COMMON platform you are ever likely to see this container run on. 

Respectfully, 
-Mark Vogt (North Aurora IL USA)

JonasVautherin, 2025-05-14:
> Respectfully,

Doesn't read like it.

> This is really really frustrating.

Providing open source tools FOR FREE and having to deal with comments like yours is frustrating as well, I'm glad you asked.

> IS ANYONE ELSE actually SUCCESSFUL with your claim "just run this" ??

Yes, many have been for years. Some have issues sometimes on systems other than Linux, and I am happy to help when they ask nicely.

I will guess that you didn't bother checking the issues and recent changes, but have you tried running the latest image? I have to ask given that you did not even bother sharing the one line you type to run it.

julianoes, 2025-05-14:
What command did you run and what's the output? Please paste it in here and maybe we can figure out what is going on.

julianoes, 2025-05-20:
@markvogt would be nice if you'd answer our questions and work through this, so we can improve it for the next person.

JonasVautherin, 2025-05-30:
@cam-s-hunt: Thanks, that's interesting! Note that px4-gazebo-headless used to work on macOS and Windows without requiring any configuration. Which is why it's not documented.

This said, we recently worked with some contributors who reported that the `latest` image worked on Windows (and maybe macOS?). Did you try it?

pawsaw, 2025-06-04:
Hi @JonasVautherin thank you for this great work. 👍🏻

I run this on a clean, fresh MacBook Pro m4 with the newest Docker Desktop setup. I cannot connect with QGC (newest).

Tl;Dr: It's been working previously on my old MacBooks but not this version on the new one. I don't know what changed, this image, docker desktop or Mac 🤷🏻‍♂️

Let me know if you can support somehow.

Kind regards
Pawel from Berlin

```
pawel@Pawels-MacBook-Pro ~ % docker run --rm -it jonasvautherin/px4-gazebo-headless:1.15.4                
Scanning dependencies of target gazebo_iris__empty
WARNING gazebo_iris__empty target DEPRECATED, please use gazebo-classic_iris__empty
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  0%] Built target git_sitl_gazebo-classic
[  0%] Performing build step for 'sitl_gazebo-classic'
[  3%] Built target std_msgs
[  6%] Built target physics_msgs
[ 11%] Built target nav_msgs
[ 11%] Built target mav_msgs
[ 13%] Built target klt_feature_tracker
[ 29%] Built target models_gen
[ 45%] Built target sensor_msgs
[ 48%] Built target OpticalFlow
[ 49%] Built target ForceVisual
[ 50%] Built target AdvancedLiftDragPlugin
[ 52%] Built target LiftDragPlugin
[ 55%] Built target gazebo_user_camera_plugin
[ 57%] Built target gazebo_groundtruth_plugin
[ 59%] Built target gazebo_lidar_plugin
[ 60%] Built target gazebo_sonar_plugin
[ 62%] Built target gazebo_random_velocity_plugin
[ 63%] Built target gazebo_opticalflow_mockup_plugin
[ 64%] Built target gazebo_camera_manager_plugin
[ 65%] Built target gazebo_opticalflow_plugin
[ 66%] Built target gazebo_airspeed_plugin
[ 70%] Built target gazebo_vision_plugin
[ 70%] Built target gazebo_uuv_plugin
[ 72%] Built target gazebo_airship_dynamics_plugin
[ 74%] Built target gazebo_gimbal_controller_plugin
[ 76%] Built target gazebo_wind_plugin
[ 76%] Built target gazebo_imu_plugin
[ 79%] Built target gazebo_mavlink_interface
[ 83%] Built target gazebo_multirotor_base_plugin
[ 83%] Built target gazebo_motor_model
[ 83%] Built target gazebo_catapult_plugin
[ 85%] Built target gazebo_magnetometer_plugin
[ 86%] Built target gazebo_usv_dynamics_plugin
[ 88%] Built target gazebo_barometer_plugin
[ 90%] Built target gazebo_parachute_plugin
[ 91%] Built target gazebo_controller_interface
[ 92%] Built target gazebo_pose_sniffer_plugin
[ 93%] Built target gazebo_aruco_plugin
[ 95%] Built target gazebo_drop_plugin
[ 96%] Built target gazebo_gst_camera_plugin
[ 97%] Built target gazebo_irlock_plugin
[ 98%] Built target gazebo_gps_plugin
[100%] Built target gazebo_video_stream_widget
[  1%] No install step for 'sitl_gazebo-classic'
[  1%] Completed 'sitl_gazebo-classic'
[  1%] Built target sitl_gazebo-classic
[  1%] Built target uorb_headers
[  2%] Built target drivers_board
[  3%] Built target events_header
[  3%] Built target flighttasks_generated
[  5%] Built target work_queue
[  5%] Built target tinybson
[  5%] Built target perf
[... 347 lines cut ...]
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>

   status        Print status for all instances
     [streams  ] Print all enabled streams

   stream        Configure the sending rate of a stream for a running instance
     [-u <val>]  Select Mavlink instance via local Network Port
     [-d <val>]  Select Mavlink instance via Serial Device
                 values: <file:dev>
     -s <val>    Mavlink stream to configure
     -r <val>    Rate in Hz (0 = turn off, -1 = set to default)

   boot_complete Enable sending of messages. (Must be) called as last step in startup script.
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-06-04/14_11_02.ulg	
INFO  [logger] Opened full log file: ./log/2025-06-04/14_11_02.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [commander] Ready for takeoff!
```

pawsaw, 2025-06-04:
Great! Adding `:latest` fixed the problem. 🎇

However, as I pulled the image (1.15.4) today it's still in the repo/hub like this. Maybe you'd like to consider a version bump and/or an updated documentation?

Again, many thanks for providing this 👍🏻

JonasVautherin, 2025-06-05:
Great that it works! I was thinking that it would come with 1.16.0, which I will release when PX4 releases it. Right now it has been in [release candidate](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.16.0-rc1) for more than a month, so I figure it should happen in the near future 😊.

JonasVautherin, 2025-08-09:
PX4 v1.16.0 has been released, and therefore px4-gazebo-headless is following. This should be fixed in v1.16.0 which is being deployed right now 🚀.

closed 2025-08-09 by JonasVautherin